espressif/esp-homekit-sdk #116: Lightbuld: fast blink on reset

leotordo, 2023-11-03:
Hello,
 on my board I've a fast blink anytime I do a reset (or on power-up).
I think it's not a issue of my code because I've the same blink ever if I switch the ESP32 in download mode.

On my board I've this logic
GPIO7=0 --> LED OFF
GPIO7=1 --> LED ON

Any idea?
Thanks
A.

leotordo, 2023-11-03:
I've the same problem with a blank device

shahpiyushv, 2023-11-03:
The description does not mention which board is being talked about and what code have you been using. Not even clear if it is related to esp-homekit-sdk. Can you please elaborate?

leotordo, 2023-11-03:
Yes, sorry I try to give you some additional informations.
The application is derived from lightbulb. I drive a n-mos with GPIO7.
I use a ESP32C3 module. The EN pin has pullup and mechanical switch to pull-down the signal.
When I release the switch (EN goes up) e see on GPIO7 a 400us pulse from 0 to VDD anche the bulb  flashes.
I find this pulse also with blank device.
My opinion is that in this time the core checks the strapping pins for download mode. May be right?
Do you think it's possible to avoid this pulse? I'm studying an HW solution, but I would prefer a FW solution.
Hope I'm clear, now.
Thank you, regards

leotordo, 2023-11-07:
I solved via HW
Thanks

closed 2023-11-07 by leotordo