espressif/esp-homekit-sdk #117: Setup Payload Generator doesn't work with Flash Download Tool

leotordo, 2023-11-08:
Hello,
 I'm trying to setup the payload generator for factory production. I generated my own nvs bin file.
On my partition table I've
`factory_nvs, data,   nvs,     0x340000,  0x6000`
so I added _0x340000 factory_nvs.bin_ at the end of flash command line and it works.

If I use flash download tool it doesn't work. The nvs isn't flashed and the program starts with hardcoded ID

[image]

Any idea?
Thanks
Leo

shahpiyushv, 2023-11-08:
As mentioned under [HomeKit Setup Configuration](https://github.com/espressif/esp-homekit-sdk#homekit-setup-configuration), have you followed this too?

To use the setup info from this image, you will have to disable the hardcoded setup code by setting CONFIG_EXAMPLE_SETUP_CODE=n (idf.py menuconfig -> Example Configuration -> Use hard-coded setup code -> disable).

leotordo, 2023-11-09:
yes, I confirm it (see image below).
I confirm also that with command line the informations are correctly downloaded

`$ esptool.py -p COM3 write_flash 0x340000 factory.bin`

the problem comes when I use Flash Download Tool. May be a bug. A the moment I proceed with command line for factory production. 
 

[image]