espressif/esp-homekit-sdk #121: How to cleanly soft reset ?

ZweiEuro, 2023-12-22:
Hi! I want to try this out with a bridge. My bridge can dynamically add endpoints. 
I would think I can do this:

start HAP
...
stop HAP
delete all accessories
re-add all now present acessories
start hap

Is this the right way of doing this? I tried rebooting and then adding/removing devices but this also causes issues with pairing. 
What would be the correct process here ?

shahpiyushv, 2024-04-26:
You can remove accessories using [hap_remove_bridged_accessory()](https://github.com/espressif/esp-homekit-sdk/blob/43bab6f68569e26c738f0273960200c809ec0ec9/components/homekit/esp_hap_core/include/hap.h#L543) and then delete them using [hap_acc_delete()](https://github.com/espressif/esp-homekit-sdk/blob/43bab6f68569e26c738f0273960200c809ec0ec9/components/homekit/esp_hap_core/include/hap.h#L553C6-L553C20) even without stopping HAP and then add them back as required.

ZweiEuro, 2024-05-16:
I understand! thank you!

closed 2024-05-16 by ZweiEuro