espressif/esp-homekit-sdk #123: HTTPD init error

baator025, 2023-12-31:
Hi. I built both my project and a clean lightbulb example with newest HomeKit SDK and IDF. During init of a program I am getting log: **HTTPD START Failed [-1]**. After that ESP manages to connect with WiFi, although I am not able to connect to my device with HomeKit. I tried using hardcoded setup code and the one flashed to partition.

shiniuniu, 2024-01-28:
Hello,[baator025](https://github.com/baator025).I'm having the same issue as well, but I've fixed it myself. You don't seem to clone the full repository, try downloading the full repository and compiling it locally. I hope this helps.

closed 2024-02-15 by shahpiyushv

Alexvrs, 2024-10-30:
Error solution for ESP-IDF v5.3.1, you need to fix the httpd configuration lines in the hap_platform_httpd.c file:

// httpd_config_t config = {
//     .task_priority  = tskIDLE_PRIORITY+5,
//     .stack_size         = CONFIG_HAP_HTTP_STACK_SIZE,
//     .server_port        = CONFIG_HAP_HTTP_SERVER_PORT,
//     .ctrl_port          = CONFIG_HAP_HTTP_CONTROL_PORT,
//     .max_open_sockets   = CONFIG_HAP_HTTP_MAX_OPEN_SOCKETS,
//     .max_uri_handlers   = CONFIG_HAP_HTTP_MAX_URI_HANDLERS,
//     .max_resp_headers   = 8,
//     .backlog_conn       = 5,
//     .lru_purge_enable   = true,
//     .recv_wait_timeout  = 5,
//     .send_wait_timeout  = 5,
//     .idle_timeout       = 60
// };

httpd_config_t config = HTTPD_DEFAULT_CONFIG();

config.stack_size = CONFIG_HAP_HTTP_STACK_SIZE;
config.server_port = CONFIG_HAP_HTTP_SERVER_PORT;
config.ctrl_port = CONFIG_HAP_HTTP_CONTROL_PORT;
config.max_open_sockets = CONFIG_HAP_HTTP_MAX_OPEN_SOCKETS;
config.max_uri_handlers = CONFIG_HAP_HTTP_MAX_URI_HANDLERS;
config.max_resp_headers = 8;
config.backlog_conn = 5;
config.lru_purge_enable = true;
config.recv_wait_timeout = 5;
config.send_wait_timeout = 5;
config.keep_alive_enable = true;
config.keep_alive_idle = 60;

shahpiyushv, 2024-10-30:
The same is part of the commit https://github.com/espressif/esp-homekit-sdk/commit/24e42654cdb458def98814283895c06340e49983 referenced above.