espressif/esp-homekit-sdk #125: Accessory Not Found

pallaviwhy, 2024-01-17:
Hi,
I have cloned homekit-sdk and downloaded the program in ESP32. I have the following displayed after giving idf.py -p COM5 monitor

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7168
load:0x40078000,len:15444
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3684
entry 0x40080614
I (31) boot: ESP-IDF v5.3-dev-1353-gb3f7e2c8a4 2nd stage bootloader
I (31) boot: compile time Jan 17 2024 12:58:49
W (32) boot: Unicore bootloader
I (36) boot: chip revision: v1.0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (78) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (85) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (92) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (100) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (107) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (115) boot:  6 factory_nvs      WiFi data        01 02 00340000 00006000
I (123) boot:  7 nvs_keys         NVS keys         01 04 00346000 00001000
I (130) boot: End of partition table
I (134) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=3f764h (259940) map
I (232) esp_image: segment 1: paddr=0005f78c vaddr=3ffbdb60 size=0088ch (  2188) load
I (233) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=fb88ch (1030284) map
I (591) esp_image: segment 3: paddr=0015b8b4 vaddr=3ffbe3ec size=050a4h ( 20644) load
I (599) esp_image: segment 4: paddr=00160960 vaddr=40080000 size=1edd0h (126416) load
I (664) boot: Loaded app from partition at offset 0x20000
I (664) boot: Disabling RNG early entropy source...
I (676) cpu_start: Unicore app
I (676) cpu_start: Single core mode
I (686) cpu_start: Pro cpu start user code
I (686) cpu_start: cpu freq: 160000000 Hz
I (686) cpu_start: Application information:
I (691) cpu_start: Project name:     lightbulb
I (696) cpu_start: App version:      bd236e7
I (701) cpu_start: Compile time:     Jan 17 2024 12:58:18
I (707) cpu_start: ELF file SHA256:  bcbdf8d63...
I (713) cpu_start: ESP-IDF:          v5.3-dev-1353-gb3f7e2c8a4
I (719) cpu_start: Min chip rev:     v0.0
I (724) cpu_start: Max chip rev:     v3.99
I (729) cpu_start: Chip rev:         v1.0
I (734) heap_init: Initializing. RAM available for dynamic allocation:
I (741) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (747) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (753) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (759) heap_init: At 3FFCAF28 len 000150D8 (84 KiB): DRAM
I (765) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (771) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (778) heap_init: At 4009EDD0 len 00001230 (4 KiB): IRAM
I (784) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (791) spi_flash: detected chip: generic
I (795) spi_flash: flash io: dio
I (799) coexist: coex firmware version: c02915e
I (804) main_task: Started on CPU0
I (804) main_task: Calling app_main()
I (814) main_task: Returned from app_main()
Keystore initialised
Accessory is not Paired with any controller
Database initialised. Accessory Device ID: F1:9B:0F:78:0C:21
HAP Initialization succeeded. Version : 4.0-bd236e7
Wi-Fi reconfiguration service not supported. Please use MFi variant of the SDK.
I (844) lightbulb: Dummy Light Driver Init.
I (854) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (864) app_hap_setup_payload: -----QR Code for HomeKit-----
I (864) app_hap_setup_payload: Scan this QR code from the Home app on iOS

  █▀▀▀▀▀█   █▀  █▀▀▀▀▀█
  █ ███ █ █▀▄▀▀ █ ███ █   
  █ ▀▀▀ █ ██▀ █ █ ▀▀▀ █
  ▀▀▀▀▀▀▀ █ █ ▀ ▀▀▀▀▀▀▀
  ▀ ▀███▀ ▄▀ ▄█▄██▀██ ▄   
  █▄█▄▀▄▀▀▀█▀▄█  ▀  ▄█▄
    ▀▀  ▀▀▄▄▀ ▀▄▀▄ ▄▄█▀   
  █▀▀▀▀▀█ ▄  ▄█▀▄█ █ ▀▄
  █ ███ █ █▀█  ▄▀▀█▀  ▀
  █ ▀▀▀ █ ▀█▄ █▄ ▄ █▀ ▄   
  ▀▀▀▀▀▀▀ ▀▀ ▀▀▀  ▀▀

I (954) app_hap_setup_payload: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-homekit-sdk/qrcode.html?data=X-HM://00527813XES32
MFi auth not supported. Falling back to HAP_MFI_AUTH_NONE
I (984) wifi:wifi driver task: 3ffd069c, prio:23, stack:6656, core=0
I (984) wifi:wifi firmware version: 33290e7
I (984) wifi:wifi certification version: v7.0
I (984) wifi:config NVS flash: enabled
I (994) wifi:config nano formating: disabled
I (994) wifi:Init data frame dynamic rx buffer num: 32
I (1004) wifi:Init static rx mgmt buffer num: 5
I (1004) wifi:Init management short buffer num: 32
I (1004) wifi:Init dynamic tx buffer num: 32
I (1014) wifi:Init static rx buffer size: 1600
I (1014) wifi:Init static rx buffer num: 10
I (1024) wifi:Init dynamic rx buffer num: 32
I (1024) wifi_init: rx ba win: 6
I (1024) wifi_init: accept mbox: 6
I (1034) wifi_init: tcpip mbox: 32
I (1034) wifi_init: udp mbox: 10
I (1044) wifi_init: tcp mbox: 6
I (1044) wifi_init: tcp tx win: 5760
I (1044) wifi_init: tcp rx win: 5760
I (1054) wifi_init: tcp mss: 1440
I (1054) wifi_init: WiFi IRAM OP enabled
I (1064) wifi_init: WiFi RX IRAM OP enabled
Setup ID: ES32
HTTPD START Failed [-1]
I (1074) wifi_prov_scheme_ble: BT memory released
I (1074) app_wifi: Already provisioned, starting Wi-Fi STA
I (1084) wifi_prov_scheme_ble: BTDM memory released
I (1094) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (1164) wifi:mode : sta (94:b9:7e:e8:f4:d0)
I (1164) wifi:enable tsf
I (3584) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (3584) wifi:state: init -> auth (b0)
I (3584) wifi:state: auth -> assoc (0)
I (3594) wifi:state: assoc -> run (10)
W (3754) wifi:[ADDBA]rx delba, code:39, delete tid:0
W (3764) wifi:[ADDBA]rx delba, code:39, delete tid:0
I (3794) wifi:connected with , aid = 2, channel 1, BW20, bssid = b4:a7:c6:b2:7e:b4
I (3794) wifi:security: WPA2-PSK, phy: bgn, rssi: -63
I (3794) wifi:pm start, type: 1

I (3794) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3814) wifi:<ba-add>idx:0 (ifx:0, b4:a7:c6:b2:7e:b4), tid:0, ssn:3, winSize:64
I (3884) wifi:dp: 2, bi: 102400, li: 4, scale listen interval from 307200 us to 409600 us
I (3884) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (4804) esp_netif_handlers: sta ip: 192.168.1.33, mask: 255.255.255.0, gw: 192.168.1.1
I (4804) app_wifi: Connected with IP Address:------
I (4974) app_wifi: Connected with IPv6 Address:----

WiFi is connected.But couldnot add accessory 
[image]

shahpiyushv, 2024-01-17:
I see an error saying "HTTPD START Failed [-1]". What happens if you just reboot the ESP32? Is this just the default lightbulb 
 example or has any custom changes?

pallaviwhy, 2024-01-17:
If i reboot the error remains the same and it is just the lightbulb example

shahpiyushv, 2024-01-18:
@pallaviwhy , seems there is some issue for esp-homekit-sdk for esp-idf master branch. The same works with release/v5.1 though. Can you try with that instead?

pallaviwhy, 2024-01-18:
I got the solution. Since i was running in esp-idf 5.3cmd version it is showing that error.
when i tried executing same code in esp-idf 5.1 cmd prompt it is working fine

Now i want to connect the hardware and check for the program execution.
void lightbulb_init(void)
{
    ESP_LOGI(TAG, "Dummy Light Driver Init.");
}
ESP_LOGI didnot get open for some reason while trying open its declaration to change the pin number.
Does  HomeKit Accessory Development Kit(ADK) is required to open it.

shahpiyushv, 2024-01-18:
The query is not clear to me. In the original logs shared, we can indeed see this line

```
I (844) lightbulb: Dummy Light Driver Init.
```

which means that this function indeed gets invoked. You can call your driver function here, instead of the dummy print. This SDK has no relation to the ADK. Both are independent.

pallaviwhy, 2024-01-18:
Yes i changed it and code is working

closed 2024-01-18 by pallaviwhy

Alexvrs, 2024-10-30:
Error solution for ESP-IDF v5.3.1, you need to fix the httpd configuration lines in the hap_platform_httpd.c file:

    // httpd_config_t config = {
    //     .task_priority  = tskIDLE_PRIORITY+5,
    //     .stack_size         = CONFIG_HAP_HTTP_STACK_SIZE,
    //     .server_port        = CONFIG_HAP_HTTP_SERVER_PORT,
    //     .ctrl_port          = CONFIG_HAP_HTTP_CONTROL_PORT,
    //     .max_open_sockets   = CONFIG_HAP_HTTP_MAX_OPEN_SOCKETS,
    //     .max_uri_handlers   = CONFIG_HAP_HTTP_MAX_URI_HANDLERS,
    //     .max_resp_headers   = 8,
    //     .backlog_conn       = 5,
    //     .lru_purge_enable   = true,
    //     .recv_wait_timeout  = 5,
    //     .send_wait_timeout  = 5,
    //     .idle_timeout       = 60
    // };

    ESP_LOGI(TAG, "Configuring HTTPD Server...");
    httpd_config_t config = HTTPD_DEFAULT_CONFIG();

    config.stack_size = CONFIG_HAP_HTTP_STACK_SIZE;
    config.server_port = CONFIG_HAP_HTTP_SERVER_PORT;
    config.ctrl_port = CONFIG_HAP_HTTP_CONTROL_PORT;
    config.max_open_sockets = CONFIG_HAP_HTTP_MAX_OPEN_SOCKETS;
    config.max_uri_handlers = CONFIG_HAP_HTTP_MAX_URI_HANDLERS;
    config.max_resp_headers = 8;
    config.backlog_conn = 5;
    config.lru_purge_enable = true;
    config.recv_wait_timeout = 5;
    config.send_wait_timeout = 5;
    config.keep_alive_enable = true;
    config.keep_alive_idle = 60;