espressif/esp-homekit-sdk #129: More than 1 accessory support without using bridge

dzungpv, 2024-03-24:
I want to add more than 1 accessory but seem now only have "bridge" and multi service in on accessory mention in: https://github.com/espressif/esp-homekit-sdk/issues/23#issuecomment-768265957
But I want multiple accessory for each service in it.
Is there any solution?
Other lib they can define multiple  accessory, only difference CID and it can add to HomeKit app: https://github.com/maximkulkin/esp-homekit-demo/issues/102#issuecomment-400006900
and this: https://github.com/maximkulkin/esp-homekit-demo/issues/102#issuecomment-662327144

shahpiyushv, 2024-03-25:
The question is not clear. A single physical device can either have
- One accessory and multiple service under it (like light, switch, etc.)
- Multiple accessories, each with own service for light or switch.
Both of these are as per the HomeKit Specifications which is what this SDK also supports.

dzungpv, 2024-03-25:
> The question is not clear. A single physical device can either have
>
>     * One accessory and multiple service under it (like light, switch, etc.)
>
>     * Multiple accessories, each with own service [...]

My purpose is: `Multiple accessories, each with own service for light or switch.` but when I add multiple accessory with 
```
    /* Add the Accessory to the HomeKit Database */
    hap_add_accessory(accessory_second);
```
for second accessory, a warning show "Primary Accessory already added. Use hap_add_bridged_accessory() instead"
code here: https://github.com/espressif/esp-homekit-sdk/blob/43bab6f68569e26c738f0273960200c809ec0ec9/components/homekit/esp_hap_core/src/esp_hap_acc.c#L384C42-L384C116

Can you give me a sample with I want to add 3 accessory on a single physical device: thermostat, switch and a lightbulb?

AramVartanyan, 2024-03-25:
Hi,

For multiple accessories implementation, you could check this example:

https://github.com/AramVartanyan/esp-homekit-sdk/blob/4e75eb0d6c024a8e10799fc4b010ebb1aca58024/examples/waterbatt/main/app_main.c#L488

Have in mind that it is not updated, but you can follow the concept.

closed 2024-04-18 by dzungpv