espressif/esp-homekit-sdk #132: esp-homekit-sdk network allocation

wirelesstag2024, 2024-04-25:
Using the lightbulb routine, using the app ESP BLE Provisioning to wire the device fails to show that the Bluetooth feature value is incorrect, and after manually changing the wiring method, it can communicate normally.
Can this demo support BLE homekit?

esp32s3 idf5.2.1

wirelesstag2024, 2024-04-25:
[image]

shahpiyushv, 2024-05-07:
I tried the same on a fresh esp32s3 board and idf v5.2.1 with default sdkconfig and I could proceed with the provisioning with ESP BLE Prov app. Do you have any custom changes? Do you see any other erros in your boot log?