espressif/esp-homekit-sdk #133: Lightbulb: save status on flash

leotordo, 2024-04-26:
Hello,
 is it possible to save status of led (on/off and % of pwm) in flash memory.
I want to recover it every time I restart the board.
Thanks
Stefano

shahpiyushv, 2024-04-26:
The SDK itself does not inherently support this, but you can store the values using [NVS APIs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_flash.html) whenever your [write callback](https://github.com/espressif/esp-homekit-sdk/blob/43bab6f68569e26c738f0273960200c809ec0ec9/examples/lightbulb/main/app_main.c#L103) is invoked or the value changes by some other means and then, on a reboot, restore it from NVS and pass it to the API which [creates the characteristics](https://github.com/espressif/esp-homekit-sdk/blob/43bab6f68569e26c738f0273960200c809ec0ec9/examples/lightbulb/main/app_main.c#L194).

leotordo, 2024-04-26:
hello.
I found this example
https://github.com/espressif/esp-idf/blob/v5.2.1/examples/storage/nvs_rw_value/main/nvs_value_example_main.c
may I write simple code where I want to save vaues?
I need to save Status (on/off) and Level (PWM)
May be useful this example?
Thanks
Stefano

leotordo, 2024-05-06:
I confirm that example above solves my request.
Bye
Stefano

closed 2024-05-06 by leotordo