espressif/esp-homekit-sdk #135: Is it possible to return the device status outside the hap_serv_set_write_cb callback function?

butaikis, 2024-07-05:
Hello. I have a task after processing a request to change the state of the device, to change the status `*(write->status)` after the callback is completed, is it possible? That is, to change the status of devices dynamically, regardless of customer requests.

shahpiyushv, 2024-07-05:
You can use [`hap_char_update_val()`](https://gitlab.espressif.cn:6688/app-frameworks/esp-homekit-sdk/-/blob/master/components/homekit/esp_hap_core/include/hap.h?ref_type=heads#L1047) any time you want to update a characteristic. Just pass it the correct characteristic handle. You can see a sample [here](https://gitlab.espressif.cn:6688/app-frameworks/esp-homekit-sdk/-/blob/master/examples/smart_outlet/main/app_main.c?ref_type=heads#L232). You can get the characteristic handle using [`hap_serv_get_char_by_uuid()`](https://gitlab.espressif.cn:6688/app-frameworks/esp-homekit-sdk/-/blob/master/components/homekit/esp_hap_core/include/hap.h?ref_type=heads#L814) or [`hap_acc_get_char_by_iid`](https://gitlab.espressif.cn:6688/app-frameworks/esp-homekit-sdk/-/blob/master/components/homekit/esp_hap_core/include/hap.h?ref_type=heads#L637) or by just keeping a copy of the characteristic during initialisation like how is done in the example above.

butaikis, 2024-07-05:
@shahpiyushv Yes, I understand that. But if my device is offline, not available, and I want to update exactly `*(write->status)` ?

shahpiyushv, 2024-07-05:
The *(write->status) is meant only to respond back the status of the write request. HomeKit data model does not have a concept of dynamically notifying such statuses. You can only report values of characteristics.

closed 2024-07-08 by butaikis