espressif/esp-homekit-sdk #138: homekit bridge Decryption error/Connection lost

smoonf, 2024-07-10:
esp32c3  **log :Decryption error/Connection lost. Marking session as invalid HomeKit Session terminated**
ios version : 17.3.1

Question1：**Does the appearance of such logs require HomeKit certification in order for them to disappear?**

shahpiyushv, 2024-07-10:
This error occurs when
- The iOS explicitly closes connection with the accessory. This typically happens when iOS wants to save power or the iOS no more needs the session
- iOS is not reachable from accessory while the accessory was trying to send an event notification
- Accessory has sent some wrong data in GET /accessories response or event notifications. For this, calling `hap_http_debug_enable()` will help get more logs

smoonf, 2024-07-10:
Thank you for your response. Below is the log obtained using hap_http_debug_enable()：

Socket fd: 51; HTTP Request POST /pair-setup
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Socket fd: 51; HTTP Request POST /pair-setup
Pair Setup M3 Received
Using pair-setup without MFi.
Pair Setup M4 Successful
Socket fd: 51; HTTP Request POST /pair-setup
Pair Setup M5 Received
Pair Setup Successful for EC3204FF-C37D-47FD-84BF-4D57FC1FB8B2
Skipping s# update for Certification requirements.
Re-announcing _hap._tcp mDNS service
Clearing Pair Setup Context
Socket fd: 51; HTTP Request POST /pair-verify
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Socket fd: 51; HTTP Request POST /pair-verify
Pair Verify M3 Received
HomeKit Session active
Pair Verify Successful for EC3204FF-C37D-47FD-84BF-4D57FC1FB8B2
Socket fd: 51; HTTP Request GET /accessories
Generating HTTP Response
{"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","characteristics":[{"iid":2,"type":"14","perms":["pw"],"ev":false,"format":"bool"},{"iid":3,"value":"Espressif","type":"20","perms":["pr"],"ev":false,"format":"string"},{"iid":4,"value":"EspBridge01","type":"21","perms":["pr"],"ev":false,"format":"string"},{"iid":5,"value":"Esp-Bridge","type":"23","perms":["pr"],"ev":false,"format":"string"},{"iid":6,"value":"001122334455","type":"30","perms":["pr"],"ev":false,"format":"string"},{"iid":7,"value":"0.9.0","type":"52","perms":["pr"],"ev":false,"format":"string"},{"iid":10,"value":"RVNQMzJIQVA=","type":"220","perms":["pr"],"ev":false,"format":"data"}]},{"iid":8,"type":"A2","characteristics":[{"iid":9,"value":"1.1.0","type":"37","perms":["pr"],"ev":false,"format":"string"}]}]},{"aid":2,"services":[{"iid":1,"type":"3E","characteristics":[{"iid":2,"type":"14","perms":["pw"],"ev":false,"format":"bool"},{"iid":3,"value":"Espressif","type":"20","perms":["pr"],"ev":false,"format":"string"},{"iid":4,"value":"EspFan01","type":"21","perms":["pr"],"ev":false,"format":"string"},{"iid":5,"value":"ESP-Fan-0","type":"23","perms":["pr"],"ev":false,"format":"string"},{"iid":6,"value":"abcdefg","type":"30","perms":["pr"],"ev":false,"format":"string"},{"iid":7,"value":"0.9.0","type":"52","perms":["pr"],"ev":false,"format":"string"}]},{"iid":8,"type":"40","characteristics":[{"iid":9,"value":false,"type":"25","perms":["pr","pw","ev"],"ev":false,"format":"bool"},{"iid":10,"value":"ESP-Fan-0","type":"23","perms":["pr"],"ev":false,"format":"string"}]}]},{"aid":3,"services":[{"iid":1,"type":"3E","characteristics":[{"iid":2,"type":"14","perms":["pw"],"ev":false,"format":"bool"},{"iid":3,"value":"Espressif","type":"20","perms":["pr"],"ev":false,"format":"string"},{"iid":4,"value":"EspFan01","type":"21","perms":["pr"],"ev":false,"format":"string"},{"iid":5,"value":"ESP-Fan-1","type":"23","perms":["pr"],"ev":false,"format":"string"},{"iid":6,"value":"abcdefg","type":"30","perms":["pr"],"ev":false,"format":"string"},{"iid":7,"value":"0.9.0","type":"52","perms":["pr"],"ev":false,"format":"string"}]},{"iid":8,"type":"40","characteristics":[{"iid":9,"value":false,"type":"25","perms":["pr","pw","ev"],"ev":false,"format":"bool"},{"iid":10,"value":"ESP-Fan-1","type":"23","perms":["pr"],"ev":false,"format":"string"}]}]},{"aid":4,"services":[{"iid":1,"type":"3E","characteristics":[{"iid":2,"type":"14","perms":["pw"],"ev":false,"format":"bool"},{"iid":3,"value":"Espressif","type":"20","perms":["pr"],"ev":false,"format":"string"},{"iid":4,"value":"EspFan01","type":"21","perms":["pr"],"ev":false,"format":"string"},{"iid":5,"value":"ESP-Fan-2","type":"23","perms":["pr"],"ev":false,"format":"string"},{"iid":6,"value":"abcdefg","type":"30","perms":["pr"],"ev":false,"format":"string"},{"iid":7,"value":"0.9.0","type":"52","perms":["pr"],"ev":false,"format":"string"}]},{"iid":8,"type":"40","characteristics":[{"iid":9,"value":false,"type":"25","perms":["pr","pw","ev"],"ev":false,"format":"bool"},{"iid":10,"value":"ESP-Fan-2","type":"23","perms":["pr"],"ev":false,"format":"string"}]}]}]}
Socket fd: 51; HTTP Request PUT /characteristics
Data Received: {"characteristics":[{"aid":4,"iid":9,"ev":true}]}
Events Enabled for aid=4 iid=9
Socket fd: 51; HTTP Request PUT /characteristics
Data Received: {"characteristics":[{"aid":2,"iid":9,"ev":true}]}
Events Enabled for aid=2 iid=9
Socket fd: 51; HTTP Request PUT /characteristics
Data Received: {"characteristics":[{"aid":3,"iid":9,"ev":true}]}
Events Enabled for aid=3 iid=9
Socket fd: 51; HTTP Request GET /characteristics?id=2.5,2.7
Generating HTTP Response
{"characteristics":[{"aid":2,"iid":5,"value":"ESP-Fan-0"},{"aid":2,"iid":7,"value":"0.9.0"}]}
Socket fd: 51; HTTP Request GET /characteristics?id=4.5,4.7
Generating HTTP Response
{"characteristics":[{"aid":4,"iid":5,"value":"ESP-Fan-2"},{"aid":4,"iid":7,"value":"0.9.0"}]}
Socket fd: 51; HTTP Request GET /characteristics?id=1.5,1.7
Generating HTTP Response
{"characteristics":[{"aid":1,"iid":5,"value":"Esp-Bridge"},{"aid":1,"iid":7,"value":"0.9.0"}]}
Socket fd: 51; HTTP Request GET /characteristics?id=3.7,3.5
Generating HTTP Response
{"characteristics":[{"aid":3,"iid":7,"value":"0.9.0"},{"aid":3,"iid":5,"value":"ESP-Fan-1"}]}
Socket fd: 51; HTTP Request GET /characteristics?id=2.10
Generating HTTP Response
{"characteristics":[{"aid":2,"iid":10,"value":"ESP-Fan-0"}]}
Socket fd: 51; HTTP Request GET /characteristics?id=3.10
Generating HTTP Response
{"characteristics":[{"aid":3,"iid":10,"value":"ESP-Fan-1"}]}
[... cut ...]
Data Received: {"characteristics":[{"aid":4,"iid":9,"value":0}]}
I (102625) HAP Bridge: Write called for Accessory ESP-Fan-2
I (102635) HAP Bridge: Received Write. Fan Off
Value Changed
Socket fd: 51; HTTP Request GET /characteristics?id=4.9,3.9,2.9
Generating HTTP Response
{"characteristics":[{"aid":4,"iid":9,"value":false},{"aid":3,"iid":9,"value":false},{"aid":2,"iid":9,"value":false}]}
Decryption error/Connection lost. Marking session as invalid
HomeKit Session terminated

**Can you help me check if there's any problem with it?**