espressif/esp-homekit-sdk #139: Problem with some services and features

butaikis, 2024-07-12:
Hello. I'm having problems with some accessories. All accessories are added through the bridge. 
First. When adding the `hap_serv_lightbulb_create()` accessory, I added the `hap_char_hue_create()` characteristic to it, but the color selection is not displayed in the Home application at all. 

The second problem with the accessory `hap_serv_thermostat_create()` when adding the application, the house writes "The accessory does not meet the requirements"
Am I forgetting to add something?

butaikis, 2024-07-16:
@shahpiyushv maybe there is a solution to the problem?

shahpiyushv, 2024-07-18:
For lightbulb, I think adding saturation along with hue may be required. For the rest, some values are outside the minValue/maxValue limits specified in HomeKit specs. These are:

```
// Target Heating Cooling State
    {
        "iid": 10,
        "value": 117,
        "type": "33",
        "perms": [
            "pr",
            "pw",
            "ev"
        ],
        "ev": false,
        "format": "uint8",
        "minValue": 0,
        "maxValue": 3,
        "minStep": 1
    }

//Target Temperature
    {
        "iid": 12,
        "value": 0,
        "type": "35",
        "perms": [
            "pr",
            "pw",
            "ev"
        ],
        "ev": false,
        "format": "float",
        "minValue": 10,
        "maxValue": 38,
        "minStep": 0.1,
        "unit": "celsius"
    }

//Cooling Threshold Temperature
    {
        "iid": 14,
        "value": 0,
        "type": "D",
        "perms": [
            "pr",
            "pw",
            "ev"
        ],
        "ev": false,
        "format": "float",
        "minValue": 10,
        "maxValue": 35,
        "minStep": 0.1,
        "unit": "celsius"
    }
```

butaikis, 2024-07-18:
@shahpiyushv Thanks, I'll try to fix it. 
Is there any examples of the HomeKit Http API in the public domain somewhere so that you can make your own implementation?

shahpiyushv, 2024-07-19:
There used to be a pdf for the non commercial version of HomeKit Accessory Specifications, but I am unable to find an official link for that right now. A crude way would be to check [these files](https://github.com/espressif/esp-homekit-sdk/tree/master/components/homekit/esp_hap_apple_profiles) in this SDK.  The [hap_apple_chars.c](https://github.com/espressif/esp-homekit-sdk/blob/master/components/homekit/esp_hap_apple_profiles/src/hap_apple_chars.c) file also has limits configured for all the characteristics supported.

butaikis, 2024-07-19:
@shahpiyushv If there is no documentation, how did you write this SDK? 
Maybe it can be downloaded from Apple (HTTP API)

closed 2024-09-26 by butaikis

shahpiyushv, 2024-09-26:
@butaikis , we have MFi developer account and so, have access to the specifications which we cannot reshare.

skanico, 2024-09-26:
@butaikis You can look for "HomeKit Accessory Protocol Specification" on Google ;-)

butaikis, 2024-09-26:
@skanico Thanks, I've already found it)