espressif/esp-homekit-sdk #140: httpd_start() don't start, has problem with create stack ESP32-S3 SPIRAM 8MB (IDFGH-13325)

MacWyznawca, 2024-07-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3 release

### Espressif SoC revision.

ESP32-S3 (QFN56) Embedded PSRAM 8MB

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom

### Power Supply used.

External 5V

### What is the expected behavior?

Start `esp-homekit-sdk` MFi, but after update IDF from 5.2.2 to 5.3 project without changes in code and settings, don't start HomeKit runtime.

### What is the actual behavior?

HomeKit rountine can't start `httpd`. Internal `httpd_os_thread_create()` can't create task. `xTaskCreatePinnedToCoreWithCaps(thread_routine, name, stacksize, arg, prio, thread, core_id, caps)`; return zero.

### Steps to reproduce.

Test `httpd` with ESP32S3 and SPIRAM

shahpiyushv, 2024-07-29:
There does not seem to be a problem with esp32s3 + idf v5.3 when I tried at my end. Are you on the latest esp-homekit-sdk, which has [this fix](https://github.com/espressif/esp-homekit-sdk/commit/24e42654cdb458def98814283895c06340e49983) for httpd start? Same commit is also available on the MFi variant of the SDK.

MacWyznawca, 2024-07-30:
Hi @shahpiyushv I force change (like on pic.) in `osal.h` file `xTaskCreatePinnedToCoreWithCaps` `caps` field to `MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT`, and on our S3 witch SPIRAM works great, but this is not elegant 😉 
[image]

MacWyznawca, 2024-07-30:
@shahpiyushv Thanks! **This fix work great** witout my not elegant changes 😉

closed 2024-07-30 by MacWyznawca