espressif/esp-homekit-sdk #142: Service button ?

butaikis, 2024-09-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I didn't find something like a button in the services. Is there any way to implement a button type device, with characteristics, click, double click, etc?

shahpiyushv, 2024-09-26:
Custom services and characteristics will work only with custom apps, not with Home app. For a button kind of service, you can use [stateless programmable switch](https://github.com/espressif/esp-homekit-sdk/blob/43bab6f68569e26c738f0273960200c809ec0ec9/components/homekit/esp_hap_apple_profiles/include/hap_apple_servs.h#L353C13-L353C58). It includes a characteristic of type `HAP_CHAR_UUID_PROGRAMMABLE_SWITCH_EVENT`, which can have following values

0: ”Single Press”
1: ”Double Press”
2: ”Long Press”

Based on your button events you can use hap_char_update_val() to report appropriate event to HomeKit. Typically this will be linked to some action from the Home app.

butaikis, 2024-09-26:
@shahpiyushv Thanks, I get it now. And what, for example, are there custom applications for custom services? 
And the second question is, if I have more commands besides "Click", "Double click", "Long press", how can I implement them?

shahpiyushv, 2024-09-26:
People building their own HomeKit compatible apps can make use of the custom services and characteristics to build custom functionality not covered by HomeKit specifications.

The Home app can only work with standard services and characteristics, that too within the limits specified for the values. 3-255 have been marked as reserved for the switch event and so the Home app wont be able to understand even if you use values 3 or beyond for any custom events.

butaikis, 2024-09-26:
@shahpiyushv Thank you. I tried to use the characteristic from 3-10 additional events began to be displayed. But the name is defined as "unknown". Is it possible to ask them?

shahpiyushv, 2024-09-26:
@butaikis ,  I do not think it will be possible. It would require spec change and a Home app update, which is very unlikely to happen. There is no way to give a name to an enum value.

butaikis, 2024-09-26:
@shahpiyushv Thanks for the quick response. I understand everything

closed 2024-09-26 by butaikis