espressif/esp-homekit-sdk #146: CONFIG_HAP_HTTP_CONTROL_PORT never use?

dzungpv, 2025-02-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I scan device with wire-shark and check current code, the HTTP_CONTROL_PORT seem never use

### Actual behavior (suspected bug)

If not using it must be remove

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Search the source code and wire-shark

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

15.3.1

### Shell

ZSH

shahpiyushv, 2025-03-04:
This is passed to the [http server ](https://github.com/espressif/esp-homekit-sdk/blob/2098b21c5d273bdd6ed7e8bfb406d4e8b5640f7a/components/homekit/esp_hap_platform/src/hap_platform_httpd.c#L33)and is used by it internally for local loopback messages, specifically to generate HomeKit Event notifications.

dzungpv, 2025-03-04:
> This is passed to the [http server ](https://github.com/espressif/esp-homekit-sdk/blob/2098b21c5d273bdd6ed7e8bfb406d4e8b5640f7a/components/homekit/esp_hap_platform/src/hap_platform_httpd.c#L33)and [...]

Thank you for the info.

closed 2025-03-04 by dzungpv