espressif/esp-homekit-sdk #148: Multiple ESP32-C6 boards show as "No Response" in Apple Home even though they are reachable

andersfischernielsen, 2025-05-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Setting up multiple ESP32-C6 boards with different `CID`s should make the devices equivalent to all other HomeKit-supported IoT devices, but they all eventually show as "No Response" in the Apple Home apps. 

### Actual behavior (suspected bug)

After setting up multiple ESP32-C6 boards with different `CID`s, both devices will initially show as responsive in Apple Home (on either macOS or iOS), but eventually all show as "No Response" in the Apple Home apps. 

The devices are still reachable through the UI (e.g. by changing the brightness of a configured light), but something prevents the device implementation from showing as available. 

The full implementation can be seen here: https://github.com/andersfischernielsen/esp-32/tree/main/ws2812b-homekit/main

### Error logs or terminal output

```shell
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3058
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4.1 2nd stage bootloader
I (23) boot: compile time May 19 2025 22:43:41
I (24) boot: chip revision: v0.1
I (24) boot: efuse block revision: v0.3
I (26) boot.esp32c6: SPI Speed      : 80MHz
I (30) boot.esp32c6: SPI Mode       : DIO
I (34) boot.esp32c6: SPI Flash Size : 2MB
I (38) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (51) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (58) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (64) boot:  2 factory          factory app      00 00 00010000 00177000
I (71) boot: End of partition table
I (74) esp_image: segment 0: paddr=00010020 vaddr=420b8020 size=29fech (172012) map
I (113) esp_image: segment 1: paddr=0003a014 vaddr=40800000 size=06004h ( 24580) load
I (119) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=b4de4h (740836) map
I (257) esp_image: segment 3: paddr=000f4e0c vaddr=40806004 size=12ba8h ( 76712) load
I (274) esp_image: segment 4: paddr=001079bc vaddr=40818bb0 size=0473ch ( 18236) load
I (284) boot: Loaded app from partition at offset 0x10000
I (285) boot: Disabling RNG early entropy source...
I (295) cpu_start: Unicore app
I (303) cpu_start: Pro cpu start user code
I (303) cpu_start: cpu freq: 160000000 Hz
I (304) app_init: Application information:
I (304) app_init: Project name:     scd41-homekit
I (308) app_init: App version:      e50e525-dirty
I (312) app_init: Compile time:     May 19 2025 22:46:42
I (317) app_init: ELF file SHA256:  10725fd9b...
I (321) app_init: ESP-IDF:          v5.4.1
I (325) efuse_init: Min chip rev:     v0.0
I (329) efuse_init: Max chip rev:     v0.99 
I (333) efuse_init: Chip rev:         v0.1
I (337) heap_init: Initializing. RAM available for dynamic allocation:
I (343) heap_init: At 40823790 len 00058E80 (355 KiB): RAM
I (348) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (353) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (360) spi_flash: detected chip: generic
I (363) spi_flash: flash io: dio
[... 262 lines cut ...]
```

### Steps to reproduce the behavior

Build and flash a simple light bulb example implementation. After a few minutes the light bulb shows as "No Response". I suspect mDNS is to blame here, but I'm not sure. Nothing seems out of the ordinary in the logs, but the device is nonetheless marked as unresponsive in Apple Home.

### Project release version

latest

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

15.4.1 (24E263)

### Shell

ZSH

andersfischernielsen, 2025-05-29:
I think this is related to the `Decryption error/Connection lost` message I'm seeing if I `idf.py erase-flash` and add the ESP in Apple Home again. Full logs: 

```
HAP Main Loop Started
mDNS initialised
Registering HomeKit web handlers
Announcing _hap._tcp mDNS service
HAP Started
I (10007) homekit: HomeKit started
I (10017) homekit: Starting LED sync task
I (10017) homekit: Started LED sync task
I (10017) main: Started HomeKit
Socket fd: 57; HTTP Request POST /pair-setup
######## Starting Pair Setup ########
Pair Setup M1 Received
Pair Setup M2 Successful
Socket fd: 57; HTTP Request POST /pair-setup
Pair Setup M3 Received
Using pair-setup without MFi.
Pair Setup M4 Successful
Socket fd: 57; HTTP Request POST /pair-setup
Pair Setup M5 Received
Pair Setup Successful for C6AF57EC-5390-4918-AFBF-F07A3DD3AD32
Skipping s# update for Certification requirements.
Re-announcing _hap._tcp mDNS service
Clearing Pair Setup Context
Socket fd: 57; HTTP Request POST /pair-verify
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Socket fd: 57; HTTP Request POST /pair-verify
Pair Verify M3 Received
HomeKit Session active
Pair Verify Successful for C6AF57EC-5390-4918-AFBF-F07A3DD3AD32
Enabling Keep-Alive on Pair Verify Session
Socket fd: 57; HTTP Request GET /accessories
Generating HTTP Response
I (30497) homekit: Read callback invoked for characteristic: 25
I (30497) homekit: Read callback invoked for characteristic: 23
I (30507) homekit: Read callback invoked for characteristic: 25
I (30507) homekit: Read callback invoked for characteristic: 8
I (30517) homekit: Read callback invoked for characteristic: 13
I (30517) homekit: Read callback invoked for characteristic: 2F
{"accessories":[{"aid":1,"services":[{"iid":1,"type":"3E","characteristics":[{"iid":2,"type":"14","perms":["pw"],"ev":false,"format":"bool"},{"iid":3,"value":"Espressif","type":"20","perms":["pr"],"ev":false,"format":"string"},{"iid":4,"value":"ESP32 Lamp","type":"21","perms":["pr"],"ev":false,"format":"string"},{"iid":5,"value":"ESP32 Lamp","type":"23","perms":["pr"],"ev":false,"format":"string"},{"iid":6,"value":"666","type":"30","perms":["pr"],"ev":false,"format":"string"},{"iid":7,"value":"1.0.1","type":"52","perms":["pr"],"ev":false,"format":"string"}]},{"iid":8,"type":"A2","characteristics":[{"iid":9,"value":"1.1.0","type":"37","perms":["pr"],"ev":false,"format":"string"}]},{"iid":10,"type":"43","characteristics":[{"iid":11,"value":true,"type":"25","perms":["pr","pw","ev"],"ev":false,"format":"bool"},{"iid":12,"value":"ESP32 Lamp","type":"23","perms":["pr"],"ev":false,"format":"string"},{"iid":13,"value":true,"type":"25","perms":["pr","pw","ev"],"ev":false,"format":"bool"},{"iid":14,"value":100,"type":"8","perms":["pr","pw","ev"],"ev":false,"format":"int","minValue":0,"maxValue":100,"minStep":1,"unit":"percentage"},{"iid":15,"value":0.00000,"type":"13","perms":["pr","pw","ev"],"ev":false,"format":"float","minValue":0.00000,"maxValue":360.00000,"minStep":1.00000,"unit":"arcdegrees"},{"iid":16,"value":0.00000,"type":"2F","perms":["pr","pw","ev"],"ev":false,"format":"float","minValue":0.00000,"maxValue":100.00000,"minStep":1.00000,"unit":"percentage"}]}]}]}
Socket fd: 58; HTTP Request POST /pair-verify
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Socket fd: 58; HTTP Request POST /pair-verify
Pair Verify M3 Received
No ctrl details found
Socket fd: 58; HTTP Request POST /pair-verify
######## Starting Pair Verify ########
Pair Verify M1 Received
Pair Verify M2 Successful
Socket fd: 58; HTTP Request POST /pair-verify
Pair Verify M3 Received
HomeKit Session active
Pair Verify Successful for C6AF57EC-5390-4918-AFBF-F07A3DD3AD32
Enabling Keep-Alive on Pair Verify Session
Socket fd: 57; HTTP Request PUT /characteristics
Data Received: {"characteristics":[{"aid":1,"iid":16,"ev":true},{"aid":1,"iid":15,"ev":true},{"aid":1,"iid":11,"ev":true},{"aid":1,"iid":11,"ev":true},{"aid":1,"iid":14,"ev":true}]}
Events Enabled for aid=1 iid=16
Events Enabled for aid=1 iid=15
Events Enabled for aid=1 iid=11
Events Enabled for aid=1 iid=11
Events Enabled for aid=1 iid=14
Socket fd: 58; HTTP Request GET /accessories
Generating HTTP Response
I (31317) homekit: Read callback invoked for characteristic: 25
[... 105 lines cut ...]
Generating HTTP Response
I (44847) homekit: Read callback invoked for characteristic: 23
{"characteristics":[{"aid":1,"iid":12,"value":"ESP32 Lamp"}]}
Socket fd: 57; HTTP Request GET /characteristics?id=1.5
Generating HTTP Response
{"characteristics":[{"aid":1,"iid":5,"value":"ESP32 Lamp"}]}
Socket fd: 57; HTTP Request PUT /characteristics
Data Received: {"characteristics":[{"aid":1,"iid":16,"ev":true},{"aid":1,"iid":13,"ev":true},{"aid":1,"iid":14,"ev":true},{"aid":1,"iid":13,"ev":true},{"aid":1,"iid":15,"ev":true}]}
Events Enabled for aid=1 iid=16
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=14
Events Enabled for aid=1 iid=13
Events Enabled for aid=1 iid=15
Socket fd: 57; HTTP Request PUT /characteristics
Data Received: {"characteristics":[{"aid":1,"iid":2,"value":true}]}
I (49577) homekit: Accessory identify called
Socket fd: 57; HTTP Request GET /characteristics?id=1.12
Generating HTTP Response
I (51897) homekit: Read callback invoked for characteristic: 23
{"characteristics":[{"aid":1,"iid":12,"value":"ESP32 Lamp"}]}
Socket fd: 57; HTTP Request GET /characteristics?id=1.5
Generating HTTP Response
{"characteristics":[{"aid":1,"iid":5,"value":"ESP32 Lamp"}]}
Decryption error/Connection lost. Marking session as invalid
```

shahpiyushv, 2025-05-30:
As per the HomeKit specs, "_A service must not have characteristics with duplicate characteristic types._" Your lightbulb service has the "On" characteristic (type 25) twice, once at iid 11 and next at iid 13. This could be one of the reasons the controller is disconnecting every now and then, or throwing errors.

andersfischernielsen, 2025-06-10:
It's been super stable for a good while now - thanks again! 😄

closed 2025-06-10 by andersfischernielsen